Andrea-Fox/peopleCounter #1: Count People at two entrance

TheMBeat, 2020-11-18:
Hello @Andrea-Fox ,

I would like to use your code to count people on a staircase. This would require using two sensors on an ESP32, as people can leave the stairs at the beginning and end. The use of two sensors is possible after a research with i2c.
Can the VL53L0X also be used instead of the VL53L1X? 

Best regards TheMBeat

Andrea-Fox, 2020-11-18:
Hi, @TheMBeat,
unfortunately I don't think that the code would work using the VL53L0X sensor, as it would require to use its own library.  Moreover, the idea behind the algorithm exploits the fact that there exist multiple zones (the region of interest) and I think that it is impossible to obtain those using a VL53L0X sensor.  
Best regards,
Andrea Fox

TheMBeat, 2020-11-18:
Hello @Andrea-Fox ,
Do the two sensors differ not only in the distance that can be measured? Therefore I thought it was also possible with the VL53L0X. 
Best regards.
TheMBeat

burton666, 2020-11-18:
No there are many differences between those two sensors, ROI (Region of interest) is one of them. You can read more [here](https://blog.st.com/vl53l1x/)

TheMBeat, 2020-11-18:
Ahh, okay.
If I have two VL53L1X, would it still be possible to count people at two doors?

burton666, 2020-11-18:
Yes, but I am pretty sure that you need one separate esp for each sensor. The sensors/code  doesnt care where you place them as they only count persons leaving or entering under/beside them. In your case the stairs would be the same as any other room and the logic will have to be executed by Home Assistant (or any other home automation system)

Andrea-Fox, 2020-11-18:
As @burton666 said, you can just place one sensor at the beginning of the stairs and one at the end and then treat the stairs as a room. Then, both sensor will register the people going in and out of the room (the stairs) and through Home Assistant you can manage the number of people in the room (and the lights in the stairs).

TheMBeat, 2020-11-19:
Hello XXX,
thanks for the quick answers and your help. I will probably order the sensors and try a little bit. 
Have nice days and stay healthy.

closed 2020-11-19 by TheMBeat