Andrea-Fox/peopleCounter #3: Wifi connection issues

burton666, 2020-11-19:
I have tried different ESP32s, different USB-cables, and tried connecting to several networks (2.4Ghz). But I only get the "Sensor online" -message. Then dots, but it never connects to any wifi. Is there any special procedure for connecting to a wifi network besides specifying the sid and password ?

Andrea-Fox, 2020-11-19:
Hi, I've observed that if you don't put into the code also the informations about the mqtt server (its IP address, the username and the pasword) for some reasons it cannot connect to the WiFi. Maybe that is the reason it doesn't connect

burton666, 2020-11-19:
Ok, I am not using any password for my mqtt server so I just used blank username/password in the sketch. I tried removing them completly from the client and then it worked.

But my next problem is that the sensor shows up in my mqttserver-logfile but after a few seconds I get this error:
`
1605794760: New client connected from 192.168.0.205 as PeopleCounterESP32Client-3a24 (p2, c1, k15).
1605794767: Client PeopleCounterESP32Client-3d85 has exceeded timeout, disconnecting.
`
Do you have any ideas of what is wrong ?

Andrea-Fox, 2020-11-19:
Maybe this [thread](https://github.com/mqttjs/MQTT.js/issues/381) might help you. However, I've never seen this kind of problem (fortunately) and I cannot understand where it comes from

burton666, 2020-11-19:
Sorry, I am stupid. I had previously just inserted the pin-headers into the esp32 but had not soldered them. The pins felt solid when picking up the esp32 but the electric connection was very sketchy 😄

kentwk, 2020-11-26:
[image]
[image]
Any Suggestion for my error msg? i try around 100 times still same error...

burton666, 2020-11-26:
Just a guess, if you are trying to connect to a 5ghz WiFi-network switch to a 2.4ghz network instead

kentwk, 2020-11-26:
> Just a guess, if you are trying to connect to a 5ghz WiFi-network switch to a 2.4ghz network instead. Otherwise I got the same error when something was wrong with the mqtt settings.

This is 2.4ghz wifi...i saw u same case with me...how u fixed this problem?

burton666, 2020-11-26:
My problem was that I did not have user/pass for my mqtt server but still had the mqtt_user/pass set to "". I solved it by removing the variables and all usage of them. So I guess your problem also is related to the mqtt server in some way.

kentwk, 2020-11-26:
> My problem was that I did not have user/pass for my mqtt server but still had the mqtt_user/pass set to "". I solved it by removing the variables and all usage of them. So I guess your problem also is related to the mqtt server in some way.

[image]
i remove related about MQTT setting, but also same error...after type sensor online then keep dot dot dot....owner pls help me figure out what is my problem....pls....

burton666, 2020-11-26:
Ok, i just deleted:
`mqtt_user` and `mqtt_pass` variables in the begining

and after that removed those variables from:
`if (client.connect(clientId.c_str(),mqtt_user,mqtt_pass))`

and ofcource your mqtt server must be configured to be able to access without user/password

Andrea-Fox, 2020-11-26:
I can only suggest you to do what @burton666 described (btw, thank you for your quick answers). The problem you're describing is related to the WiFi.h file and unfortunately I cannot know what causes it.  
From your pictures I see that you've setted as device "SparkFun ESP32 Thing", while I've always used ESP32 Dev Module ; maybe your issue is related to that

sotidis, 2020-11-26:
WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) 
  {
    delay(5000);
    Serial.print(".");
    WiFi.begin(ssid, password);

Try to increase  delay, it helped me.

kentwk, 2020-11-27:
Finally success to connect MQTT, thanks everyone, i just follow @sotidis solution to fix it. i still have one problem about automation.yaml
[image]
[image]

kentwk, 2020-11-27:
Finally success to connect MQTT, thanks everyone, i just follow @sotidis solution to fix it.

closed 2020-11-27 by Andrea-Fox