Andrea-Fox/peopleCounter #4: People in the room now: 111227

sotidis, 2020-11-19:
Hello . is it ok?

: warning: 'RIGHT' defined but not used [-Wunused-variable]
 static int RIGHT = 1;
            ^

One person has entered in the room. People in the room now: 111227
One person has exited the room. People in the room now: 71227

Andrea-Fox, 2020-11-19:
No, there is definitely something wrong. Later this day I will update the code to eliminate this kind of behavior, however my advice is to use this code only for finding when someone is entering (or exiting) the room and keep count of the people inside the room using, for example, Home Assistant

sotidis, 2020-11-19:
Thanks ! i use iobroker ,if the value People in the room now: "-" 111227,i use iobroker,  does not count correctly.

Andrea-Fox, 2020-11-20:
I've edited the code and eliminated the lines of code that produced that output. Thank you for noticing that

closed 2020-11-20 by Andrea-Fox