Andrea-Fox/peopleCounter #6: Updated ST algorithm V2

markjones89, 2021-01-07:
Hi @Andrea-Fox 

Thanks for this project, I noticed on ST's website that they have released version 2 of there code, from what I can see they have updated the code to include a buffer of measurements to feed into the counting algorithm for better accuracy. This might be of some interest for your project. 

https://www.st.com/en/embedded-software/stsw-img010.html

Andrea-Fox, 2021-01-07:
Thank you for the suggestion! I will give a look in the future and see if I'm able to create a similar updated version also for esp32/esp8266

itekhi, 2022-10-19:
Would love to see the V2 update :)
Do you have a V1 example of their code? I could compare them to see what's new.

Andrea-Fox, 2022-11-03:
Hi, I've added in the original_code folder their V1 code which is currently not available on their website

closed 2022-11-03 by Andrea-Fox