Andrea-Fox/peopleCounter #8: Esphome integration

DutchDeffy, 2021-01-26:
Do you know esphome? It has seamless integration with home assistant. I'm busy on a way to integrate the people counting people concept with the VL53L1X sensor to Esphome. Don't know if it's working this way but here's is the repo: https://github.com/DutchDeffy/esphome_custom_component_VL53L1X_people_counter.

Andrea-Fox, 2021-01-26:
Unfortunately I discovered esphome only a few weeks after creating this project :)  
However I'm planning to integrate this idea into that project, as it is much better and would probably reach an higher number of people

DutchDeffy, 2021-02-05:
There’s already a pull request for adding the sensor to Esphome as a component which passsd all the checks and is to be reviewed and checked. The code doesn’t do the algorithm and people counting logic yet. Currently it does do the distance measuring part, which is already really brilliant. Maybe it could help you out. The PR is referenced by #1447 and #916 for the docs.

Andrea-Fox, 2021-02-05:
I will definitely look into it. Thank you for the informations

closed 2021-02-05 by Andrea-Fox