Andrea-Fox/peopleCounter #9: High power consumption -> not possible to run via battery

LC7894, 2021-01-26:
I measured the power consumption of the sensor and found it to use (on average) 80mA for a setup with an ES8266 NodeMCU. Therefore, using a battery as power source is not suitable except you want to change the battery in very small intervals. This is a problem when there is no outlet nearby or you don't want to use a cord. According to the docs of the sensor it should be possible to use the deep sleep mode of the ESP and wake it up when someone passes the sensor:

[image]

I just leave this here as a suggestion for further development of the algorithm 🙂 (Maybe I am able to find a solution on my own, but I have to learn a lot about ESP programing until I reach this point)

Andrea-Fox, 2021-01-26:
I know that the consumption are very high, however I wanted to always have the sensor connected to the wifi, in order to minimize the time that passes between passing under the sensor and turning on (or off) the light.  
I know that turning on and off the wifi should be almost immediate, however I think that even at that point the consumption given by the VL53L1X sensor and the ESP itself (with wifi off) make it impossible to connect the whole device to a battery and have a long battery life. However, if you manage to create a sensor that works with battery and has a long battery life (at least a few months I would say), let me know

LC7894, 2021-01-26:
The delay between passing the sensor and turning on/off the light is definitely a disadvantage when it comes to turning off the WiFi to save some battery. In addition (as you said) the consumption of the sensor + ESP itself with wifi off is still high. So that is another challenge.

I'll let you know if I managed to create a sensor with a long battery life (long = > 6 months) and a delay that is acceptable. But I see some challenges here so let's see.

avdept, 2021-03-31:
How about to use zigbee instead of wifi? Should use less power

Andrea-Fox, 2021-04-01:
Is there a way to use zigbee with the esp32/esp8266?

profx5015, 2021-05-19:
> Is there a way to use zigbee with the esp32/esp8266?

Here is a library to connect atmega328 with cc2530.
https://github.com/VertorixAU/Vertorix_VT1100_Mini