Andrea-Fox/peopleCounter #16: Possible enhancement with 4 ROIs

markjones89, 2021-02-26:
Hi @Andrea-Fox and @noxhirsch, Thanks so much for what you have done here!

I have been thinking of possible ways to make the sensor more accurate on wider doorways.

My thinking behind this is to use more than 2 ROIs but rather use 4 zones of 8x8 two would be in the front zone and two in the back zone. ( Note I haven't had much success with an ROI set to 8 x 16 but 8x8 works well but sometimes misses counts if you approach a doorway from an angle)

I am working on trying to implement this but would be interested in your opinion on the idea?

I was considering two possible ways of implementing this.

option 1: feed the two values from the front zone into the counting algorithm at the same time and the same for the back zone
option 2: feed the lowest measurement reading from the two zones in the front and back into the algorithm.

Thank you
Mark

LC7894, 2021-02-26:
Hi @markjones89,

I also have the issue that the sensor is not always accurate on wider doorways. Until now I was not able to find a more reliable solution.

Your idea with the four ROIs might be a solution to the problem, but unfortunately the VL53L1X sensor is only capable of using two zones and not four (hardware restriction). For this reason your approach will not work in my eyes (correct me if I'm wrong).

noxhirsch, 2021-02-26:
Maybe for wider doorways it's easier to mount them on one side? I did this with my old VL53L0X sensors that I used as simple motion sensors:
[image]
[image]
As far as I remember other people on the Discord server also tried that.

Andrea-Fox, 2021-02-26:
@LC7894 are you sure that you cannot use more than two zones? Because I've never tried and I had no idea of this limitation

@markjones89 I think that the idea should work, however, I'm not sure it would be as precise. The sensor is only able to measure the distance one ROI at a time and each measurement takes up to 100 ms (now it's 50 ms for each measurement and 50 of delay between two different measurements; I've tried to reduce this time but it is not possible, unfortunately). Having 4 zones might not allow being able to detect someone passing under it properly due to the long time that passes between one measurement and the other.
I can suggest you to try putting the sensor on the side of your door; by doing so the width of the door won't be an issue (unless you have a door wider than 4 meters

markjones89, 2021-02-26:
Thank you all, much appreciated the quick feed back. I've also had the idea to make an new version with 2 sensors. I've gone as far as designing a pcb. It's got a few errors in the design but would be happy to share the files. [image]

markjones89, 2021-02-27:
Maybe a possible solution for the timing for 4 zones could be to alternate between two front zones Ans two back zones. 
Eg zone 1,3 are front zones 2,4 are back zones.

The sensor first checks zone 1 and 2 then swaps to zone 3 and 4, this would still use the same timing bit could check a larger area. I'm going to test it out and see if it makes an improvement.

Thanks all for your comments.
[image]

Andrea-Fox, 2021-02-27:
I'm not entirely sure of it is going to work, however if you obtain good results let us know. I'm still convinced that putting the sensor on the side of the door is a better idea in this case

hoangtruong2412, 2021-07-30:
hi, is there any specific formula to calculate detection area? I installed the sensor at the door level, my doorway is about 85cm wide. but if I go too close to the sensor or too far it doesn't recognize it, I have to go in the middle of the sensor door frame to recognize it relatively well. 10 times passing but 4 to 5 times the sensor does not recognize it is a sad thing.