Andrea-Fox/peopleCounter #18: ESPHome compatible sketch?

drageloth, 2021-02-28:
Hello,

Sorry for declaring it as an issue, it's more of a question. Will there be a file available for ESPHome? It's way easier, especially for HomeAssistant to integrate automatically. Thanks!

Andrea-Fox, 2021-02-28:
Unfortunately I'm not able to integrate this idea inside ESPHome. Me (and other people) have tried to create an integration, as we understand that it would be much easier and it would reach a broader audience, however we found that creating an ESPHome integration is very complex.
You can find more informations [here](https://github.com/esphome/esphome/pull/1447) and if you are able to create the integration (or know someone who is able to), it would help a lot

drageloth, 2021-03-01:
I haven't the slightest idea to be honest, but could [https://github.com/diplix/esphome-configuration](url) help with that? Seems it's already for ESPHome, though it will need some files directly imported to the ESPHome folder. But it's a start. The problem with diplix is some false negatives with people with different heights (I'm getting regocgnised at all times, wife at 70%)

Andrea-Fox, 2021-03-01:
If I'm understanding correctly, the repository you linked uses the VL53L1X sensor only to count people sitting on a bathroom. Am I correct? Because the code in this repository is made for counting people entering (or getting out) of a room

drageloth, 2021-03-01:
> If I'm understanding correctly, the repository you linked uses the VL53L1X sensor only to count people sitting on a bathroom. Am I correct? Because the code in this repository is made for counting people entering (or getting out) of a room

It has this option too, but the main one is people counting sensor. The sketch exposes a multisensor. Picture attached to see for the counting sensor.
[image]
In main terms it's good. It can track entry and exit(but only exposes a number of people inside, not actual entry or exit), it can track if someone is standing on the door and if the person doesn't enter, it doesn't count as entry and has some other bits and pieces (distance sensor, restart etc)
The only big downside I've found is with the wife as she's 1.75m(I'm 1.93m) and sometimes the sensor doesn't count either exit or entry, making the people_count sensor wrong(eg 1 person inside when noone is).

Andrea-Fox, 2021-03-01:
What I have observed is that in cases such as the one you explained, where only taller people are identified, the error is due to a wrong value of a threshold that defines if someone is passing under the door. It probably is higher than it should be.
However, thank you for making me discover that repository. It might be useful in the future

drageloth, 2021-03-01:
Thanks too! Will keep an eye on this repo too!

closed 2021-03-01 by drageloth