Andrea-Fox/peopleCounter #19: Pls help check my hardware support anot?

kentwk, 2021-04-10:
[image]
[image]
[image]

Pls help check my hardware support anot? Because I walk from which way result keep zero.

Andrea-Fox, 2021-04-11:
The connections seem correct. Maybe the issue is the sensor. First of all, is it a VL53L1X sensor? If so, in general there is some sort of cover on the sensor. Try removing that.  
If then it still doesn't work, try connecting it to a 5v pin

kentwk, 2021-04-12:
> The connections seem correct. Maybe the issue is the sensor. First of all, is it a VL53L1X sensor? If so, in general there is some sort of cover on the sensor. Try removing that.
> If then it still doesn't work, try connecting it to a 5v pin

1)Yes...is VL53L1X-V2
1)Soft cover removed
2)Tried use 5v pin...not working

Do u mind show your sensor?

Andrea-Fox, 2021-04-18:
Then I really don't know which is the problem. 
Maybe the sensor is defective

kentwk, 2021-04-18:
> Then I really don't know which is the problem.
> Maybe the sensor is defective

but i have tried few sensor also same...izzit u sensor same with me?if working izzit serial monitor result 0 will change 1 then 2?

Andrea-Fox, 2021-04-22:
Currently I'm using a sensor from Pololu, which is not the one you're using.

When working, in the signal monitor you should read the distance measured by the sensor (in your case, unfortunately, always equal to 0)

kentwk, 2021-05-02:
> Currently I'm using a sensor from Pololu, which is not the one you're using.
> 
> When working, in the signal monitor you should read the distance measured by the sensor (in your case, unfortunately, always equal to 0)

Do u mind screen shot your serial monitor?

kentwk, 2021-05-17:
> Currently I'm using a sensor from Pololu, which is not the one you're using.
> 
> When working, in the signal monitor you should read the distance measured by the sensor (in your case, unfortunately, always equal to 0)

After soldering sensors working.TQ
i use manual_calibration can count people, when using autocalibration no working...may i noe using autocalibration what mqtt topic should i write?

Andrea-Fox, 2021-05-17:
If you are using the code in the file called `people_counter_esp32.ino` or the one in `people_counter_esp8266.ino`, you have to subscribe to the topic `people_counter/NAME_OF_THE_ROOM/counter`, while if you want to read the distance you have to subscribe to the topic `people_counter/NAME_OF_THE_ROOM/distance`.
To specify the name of the room, you have to change the code in the line indicated in the file

sanderlv, 2021-05-18:
same issue maybe here...
[image]

successfully flashed but not connecting to wifi and no serial logging in arduino IDE? I do get some on ither serial monitor
```
Using 'COM16' as serial port.
Showing logs:
[19:39:12]ets Jun  8 2016 00:22:57
[19:39:12]
[19:39:12]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[19:39:12]configsip: 0, SPIWP:0xee
[19:39:12]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[19:39:12]mode:DIO, clock div:1
[19:39:12]load:0x3fff0018,len:4
[19:39:12]load:0x3fff001c,len:1216
[19:39:12]ho 0 tail 12 room 4
[19:39:12]load:0x40078000,len:10944
[19:39:12]load:0x40080400,len:6388
[19:39:12]entry 0x400806b4
[19:39:12]E (58) psram: PSRAM ID read error: 0xffffffff
[19:39:12][19:39:12]
[19:39:12]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[19:39:12]configsip: 0, SPIWP:0xee
[19:39:12]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[19:39:12]mode:DIO, clock div:1
[19:39:12]load:0x3fff0018,len:4
[19:39:12]load:0x3fff001c,len:1216
[19:39:12]ho 0 tail 12 room 4
[19:39:12]load:0x40078000,len:10944
[19:39:12]load:0x40080400,len:6388
[19:39:12]entry 0x400806b4
[19:39:12]E (59) psram: PSRAM ID read error: 0xffffffff
```

kentwk, 2021-05-19:
> same issue maybe here...
> [image]
>
> successfully flashed but not connecting to wifi and no serial logging in arduino IDE? I do get some on ither serial monitor
>
> ```
> Using 'COM16' as serial port. [...]

Maybe u can check few things
1)void setup_wifi() 
{
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) 
  {
    delay(5000);
    Serial.print(".");
    WiFi.begin(ssid, password);
  }
}

***Delay from 500 change to 5000***

2)Serial monitor set 9600baud
[image]

sanderlv, 2021-05-19:
Serial monitor was 9600 👍 but I can not get any output?
[image]

I changed the 500 to 5000 and it comes alive now... will proceed further with homeassitsant integration etc.

thanks!!

kentwk, 2021-05-19:
Make sure to fill in correct information.

[image]

sanderlv, 2021-05-19:
Yes, that is correct because I am getting mqtt information:
[image]

Oh... for my sensor orientation? Do you happen to know like below:
https://github.com/Andrea-Fox/peopleCounter/blob/master/autocalibration/sensor_orientation.png

I have a VL53L1X

kentwk, 2021-05-19:
> Yes, that is correct because I am getting mqtt information:
> [image]
> 
> Oh... for my sensor orientation? Do you happen to know like below:
> https://github.com/Andrea-Fox/peopleCounter/blob/master/autocalibration/sensor_orientation.png
> 
> I have a VL53L1X

no...I use the manual calibration counting people is working...if using auto calibration no working, no matter what i change MQTT topic also same.

sanderlv, 2021-05-19:
for an ESP32 what would I have to use?

https://github.com/Andrea-Fox/peopleCounter/tree/master/manual_calibration

kentwk, 2021-05-19:
peopleCounter.ino

sanderlv, 2021-05-19:
Hm... that looks complicated for now. i want to try with auto first. I do not see any change on the mqtt tpoic people_counter/tof_people/counter... I do see continuous change on the people_counter/tof_people_ard/distance topic... any tips/suggestion?

kentwk, 2021-05-19:
**Manual Calibration Setting**
1)double-check u serial monitor working anot...if same with my screenshot mean working.then u can next step
[image]

2)Go Homeassistant-Congfiguration-Helpers-Add Helper
[image]

3)Go Homeassistant-Files Editor-automations.yaml-add below these script
[image]

4)Done
[image]

kentwk, 2021-05-19:
u using manual calibration or auto calibration?

**Follow this Topic**
topic: "peopleCounter/serialdata/tx"

sanderlv, 2021-05-19:
Auto

kentwk, 2021-05-19:
> Auto

u try using manual...because auto calibration also no working for me.

sanderlv, 2021-05-19:
Sorry but I have read that and I do not know how to start with that manual calibrarion

kentwk, 2021-05-19:
> Sorry but I have read that and I do not know how to start with that manual calibrarion

just flash the manual calibration files...then follow my step.

kentwk, 2021-05-20:
u forgot change wifi delay?

void setup_wifi()
{
WiFi.begin(ssid, password);
while (WiFi.status() != WL_CONNECTED)
{
delay(5000);
Serial.print(".");
WiFi.begin(ssid, password);
}
}

Delay from 500 change to 5000

sanderlv, 2021-05-20:
Don't know how that happened... I did thought I checked it, but probably not well enough.. I get now:
[image]

Andrea-Fox, 2021-05-20:
Hi, what you are seeing in your last message is the correct behavior. The topic in `peopleCounter/serialdata` is updated only whenever a passage is detected. 
A passage is detected whenever the array of 4 numbers (the one which is in general 0000) is either 0132 or 0231

kentwk, 2021-06-04:
Let me conclude my topic...just my soldering problem. TQ

closed 2021-06-04 by kentwk