Andrea-Fox/peopleCounter #21: counter = 1, but never 2

sanderlv, 2021-05-19:
Starting a new topic here.

I used:
people_counter_esp32.ino
the name of my sensor is `tof_people_ard`

Changed the value 500 to 5000 for wifi timeout.

The ESP32 is succesfully conected to WiFi and MQTT. I can see the topic realtime changing for:
`people_counter/tof_people_ard/distance`

I do not see any updates for the counter in topic:
`people_counter/tof_people_ard/counter`

It did work 1 time hower.

[image]

In above image you can see the MQTT realtime going and on some moment I place my hand before the sensor.

the standard config as in the ino is in my compiled upload. And no matter what I try to place the sensor it counts 2... so the home assistant counter is never updated.

Any help appreciated!

Andrea-Fox, 2021-05-20:
The array of 4 elements is the one which detects if a passage is detected. 
The counter topic sends a message (either 1 or 2) only whenever the array has values 0132 or 0231. This can happen only when this happens:
1) something is detected in zone A but nothing is in zone B (at this moment the array has value 0100 or 0200)
2) something is detected in both zones  (0130 or 0230)
3) something is detected in zone B but nothing is in zone A (0132 or 0231)
4) nothing is detected in both zones

If you want to simulate the passage with your hand you have to move slowly (the same speed when you are walking). Moreover, try changing the value of the threshold percentage, as it might be too high for your case

kentwk, 2021-06-06:
// this value has to be true if you don't need to compute the threshold every time the device is turned on
static bool save_calibration_result = false;

**Default value is false...try set to true, then will working.**